dexsuite/dex-retargeting #2: hand root position

JiaheXu, 2023-09-12:
Hi @yzqin 

For the current code, does it produce the 3D position estimation of the hand? I am not sure if my understanding is correct, the media-pipe produces the 3D joint positions in hand's joint 0 frame? What about the wrist rotation matric, is it in camera frame?

yzqin, 2023-09-12:
Yes @JiaheXu you are correct.

The MediaPipe 3d joint positions are local, represented in the hand wrist frame (joint 0 frame). For the rotation matrix computed via [this function](https://github.com/dexsuite/dex-retargeting/blob/32da0d0381c891203b6a7ce709221f6a1b0891a1/example/vector_retargeting/single_hand_detector.py#L110), it is in camera frame. 

For hand detection, this repo currently only provides part of the code in [AnyTeleop](http://anyteleop.com). It does not include the the detection code to estimate the position of wrist in the camera frame, only orientation. Position estimation code is much more complex and will be release later from NVIDIA.

closed 2023-11-07 by yzqin

BIT-wh, 2024-06-26:
Hi @yzqin ! Sorry to bother you again. 

May I ask if NVIDIA has now released the code for the wrist position estimation? Only RGB or RGB-D is ok.

BIT-wh, 2024-06-26:
Hi @yzqin ! Sorry to bother you again.

May I ask if NVIDIA has now released the code? You use [FoundationPose] for the wrist position estimation?