dexsuite/dex-retargeting #4: Failed loading a new robot hand model into RetargetingConfig.

RyanJiao, 2023-11-06:
Hi @yzqin

Thank you so much for your amazing work. 
While loading a new robot hand model(similar to Allegro Hand) using this code: 
```
retargeting = RetargetingConfig.load_from_file('/home/ryanjiao/RoboticManipulator/dex-retargeting/dex_retargeting/configs/teleop/leap_hand_right.yml').build()
 ```
 I encountered an error as follow:
```
/home/ryanjiao/anaconda3/envs/retgt/lib/python3.8/site-packages/numpy/lib/twodim_base.py:295: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray.
  v = asanyarray(v)
Traceback (most recent call last):
  File "/home/ryanjiao/RoboticManipulator/dex-retargeting/example/vector_retargeting/real_time_mujoco.py", line 106, in <module>
    main()
  File "/home/ryanjiao/RoboticManipulator/dex-retargeting/example/vector_retargeting/real_time_mujoco.py", line 37, in main
    retargeting = RetargetingConfig.load_from_file('/home/ryanjiao/RoboticManipulator/dex-retargeting/dex_retargeting/configs/teleop/leap_hand_right.yml').build()
  File "/home/ryanjiao/RoboticManipulator/dex-retargeting/dex_retargeting/retargeting_config.py", line 137, in build
    sapien_model = SAPIENKinematicsModelStandalone(
  File "/home/ryanjiao/RoboticManipulator/dex-retargeting/dex_retargeting/optimizer_utils.py", line 109, in __init__
    builder = loader.load_file_as_articulation_builder(urdf_path)
IndexError: stof
```
I created an yaml file for this robot hand model and the content of the yaml file is almost the same as the yaml file for Allegro Hand.
And my URDF work fine in Mujoco environment. But the `meshes` in my URDF is using `.stl` instead of `.mtl+.obj`. I wonder whether this would cause the error?

[image]

RyanJiao, 2023-11-06:
I change .stl to .obj and it work!
Sorry for bothering you.

closed 2023-11-06 by RyanJiao

tianhaowuhz, 2023-12-20:
> I change .stl to .obj and it work! Sorry for bothering you.

Hi, I am also try to use the leap hand model, but even though I change the stl to obj, I still meet same issue, how do you convert the stl? Or do you made any other changes to urdf?

yzqin, 2023-12-21:
Check issue https://github.com/dexsuite/dex-retargeting/issues/7 with our provided example on leap hand