dexsuite/dex-retargeting #5: I'm having the following issue running your code.

SshiJwu, 2023-11-11:
Thanks for your useful work!
I'm having the following issue running your code: 
python detect_from_video.py   --robot-name allegro   --video-path data/human_hand_video.mp4   --retargeting-type vector   --hand-type right   --output-path data/allegro_joints.pkl 
Traceback (most recent call last):
  File "detect_from_video.py", line 9, in <module>
    from dex_retargeting.retargeting_config import RetargetingConfig
  File "/public/home/wushj12023/dex-retargeting-main/dex_retargeting/retargeting_config.py", line 1, in <module>
    import sapien.core as sapien
  File "/public/home/wushj12023/.conda/envs/retarget/lib/python3.8/site-packages/sapien/__init__.py", line 9, in <module>
    from sapien import core, sensor, asset, example, utils
  File "/public/home/wushj12023/.conda/envs/retarget/lib/python3.8/site-packages/sapien/core/__init__.py", line 3, in <module>
    from .pysapien import *
ImportError: libvulkan.so.1: cannot open shared object file: No such file or directory
I run your code on the GPU server, so I can't execute the sudo command. How do I fix this?

yzqin, 2023-12-13:
I think this is a issue of your `sapien` installation.  Can you create a new conda environment and run python with `import sapien` to check what is the root cause for that. The `libvulkan.so` should be provided in the sapien pip package. 

To further debug this issue, can you provide the results of `pip list | grep sapien` so that we know what is your sapien version?

SshiJwu, 2023-12-13:
I installed Sapien on my local computer. It may be caused by servers that do not support non graphical interfaces.

closed 2023-12-21 by yzqin