dexsuite/dex-retargeting #6: Using Isaac sim to reproduce your code.

SshiJwu, 2023-12-13:
Hi,yuzhe!
I am using Isaac Sim to reproduce your code.But I encountered a problem during the reproduction process.I really hope you can give me some help to solve this problem. The problem mainly arises in the calculation of Jacobian matrices for specific links. As shown in this image:
[image]
 I use the API '**articulation_view.get_jacobians()'** and the  shape of returned result is (33,6,30). I tried slicing it to obtain the Jacobian matrix for a specific link. But I found that the result calculated by your code is inconsistent with the result returned. Given the position of the joint, how do I calculate the Jacobian matrix for a specific body? I really hope you can give me some help to solve this problem.

yzqin, 2023-12-13:
Hi @SshiJwu ,

I'm not deeply versed in IsaacSim, yet from my understanding and previous exposure to IsaacGym, I suspect that the discrepancy you're observing between SAPIEN and IsaacSim stems from their distinct interpretations of the Jacobian. In SAPIEN, the Jacobian is related to the spatial twist, whereas in Isaac, it pertains to cartesian velocity combined with angular velocity. Typically, when rotational dynamics are negligible, these Jacobsian representations might be seen as interchangeable. However, complications can arise due to the varying joint sequencing as interpreted by different URDF parsers. It's crucial to ensure that you maintain a consistent joint order across your various applications.

Moreover, I'm intrigued as to why there's a necessity to recode for IsaacSim. Should you be utilizing IsaacSim for simulations, it's feasible to apply our existing codebase without alterations, despite the embedded SAPIEN simulator. The retargeting code we've developed is capable of calculating the necessary joint angles, which you can then input directly into the IsaacSim environment. I'm of the opinion that redeveloping this codebase to employ an alternative Jacobian computation offers no clear advantage.

yzqin, 2023-12-13:
@SshiJwu 

Regarding your query about the out-of-bound value, it seems likely that the problem arises from the `last_qpos` input in `self.optimize(objective_fn, last_qpos)` being beyond the allowable range for that particular joint. This could result from an unsuccessful optimization process in the previous iteration, potentially due to alterations in the Jacobian calculation within the original codebase.

SshiJwu, 2023-12-14:
Yes. The order of the joints returned by the Isaac SIM is inconsistent with that of SAPIEN.
In Isaac Sim, it is ['WRJ1', 'FFJ4', 'LFJ5', 'MFJ4', 'RFJ4', 'THJ5', 'WRJ2', 'FFJ3', 'LFJ4', 'MFJ3', 'RFJ3', 'THJ4', 'FFJ2', 'LFJ3', 'MFJ2', 'RFJ2', 'THJ3', 'FFJ1', 'LFJ2', 'MFJ1', 'RFJ1', 'THJ2', 'LFJ1', 'THJ1'], but ['WRJ2', 'WRJ1', 'FFJ4', 'FFJ3', 'FFJ2', 'FFJ1', 'MFJ4', 'MFJ3', 'MFJ2', 'MFJ1', 'RFJ4', 'RFJ3', 'RFJ2', 'RFJ1', 'LFJ5','LFJ4','LFJ3','LFJ2','LFJ1','THJ5','THJ4','THJ3','THJ2','THJ1'] in SAPIEN.
I changed the code a bit,as follows:

> 
    def _get_objective_function(self, target_pos: np.ndarray, fixed_qpos: np.ndarray, last_qpos:np.ndarray):
        qpos = np.zeros(self.robot_dof)
        qpos[self.fixed_joint_indices] = fixed_qpos
        torch_target_pos = torch.as_tensor(target_pos)
        torch_target_pos.requires_grad_(False)

        def objective(x: np.ndarray, grad: np.ndarray):
            qpos[self.target_joint_indices] = x

            all_frame_names = self.kinematics_solver.get_all_frame_names()
            target_frame_name_index = [self.kinematics_solver.get_all_frame_names().index(body_name) for body_name in self.body_names]

            positions = []
            base_jacobian = []
            for index in target_frame_name_index:
                position , rotation  = self.kinematics_solver.compute_forward_kinematics(all_frame_names[index],qpos)
                base_frame_name = self.kinematics_solver.kinematics.jacobian(qpos,all_frame_names[index])[:3,:]
                positions.append(position)
                base_jacobian.append(base_frame_name)
            body_pos = np.array(positions)

            torch_body_pos = torch.as_tensor(body_pos)
            torch_body_pos.requires_grad_()

            huber_distance = self.huber_loss(torch_body_pos , torch_target_pos)
            result = huber_distance.cpu().detach().item()

            if grad.size>0:

                jacobians = np.stack(base_jacobian, axis=0)

                huber_distance.backward()
                grad_pos = torch_body_pos.grad.cpu().numpy()[:, None, :]

                grad_qpos = np.matmul(grad_pos, np.array(jacobians))
                grad_qpos = grad_qpos.mean(1).sum(0)
                grad_qpos += 2 * self.norm_delta * (x - last_qpos)
                grad[:] = grad_qpos[:]

            return result
        
        return objective
In this case, I looked at the calculation of the Jacobian matrix, and it is almost unanimously possible

yzqin, 2023-12-15:
Consider this library as an opaque utility that carries out the necessary retargeting calculations for your application. Once you receive the retargeting data from this library, you can readily apply it to simulations or visualizations within IsaacSim. There's no need for you to delve into the computational details managed by SAPIEN.

Regarding the Jacobian, I'm not certain of the specific definition used in Isaac Sim. The function `compute_single_link_local_jacobian` calculates the **spatial twist Jacobian** for a given link as defined in the URDF. It produces a matrix with the dimensions `7*n`, where `n` represents the degrees of freedom (DOF) of the robot.

SshiJwu, 2023-12-17:
Thanks,yuzhe!

SshiJwu, 2023-12-24:
I also have a question about the left and right hands. What is the difference between the left and right dexterous  hand URDF? Can I retarget the data from the left hand to the URDF of the right hand.

SshiJwu, 2023-12-24:
I've found that if I retarget the data from the left hand to URDF of the right hand, I get a big shift between the hand and the object.
[image]

SshiJwu, 2023-12-25:
Why did I change the hand.type of DexYCB to left? How can the relative position of the hand and the object be so different after rendering?
[image]

yzqin, 2024-01-03:
It does not make sense to retarget the motion from right hand to left hand due to the huge morphology gap. Always retargeting left human hand motion to a left hand robot and vice versa.

SshiJwu, 2024-01-07:
There is another question that has been bothering me for a long time, in order to allow the dexterous hand to move freely in space, you added six free joints.  The translation_range is (-1,1),and the rotation_range is (-pi,pi). Why is the range of translation this? Since space is infinite, I can use my dexterous hands to reach anywhere in the space.

yzqin, 2024-01-07:
You can tune this value based on your applications. For example, you can use a very large limit for translation to make the hand more freely inside the space. The `(-1, 1)` is an example of motion range, there are no specific meaning of these numbers,

closed 2024-04-19 by yzqin