dexsuite/dex-retargeting #7: Faild to load Leap Hand model

tianhaowuhz, 2023-12-20:
I am trying to use the leap hand model, the model I use is  from https://github.com/dexsuite/dex-urdf. However, it stills failed to load.
[image]
Is there anything I should notice for using new robot model?

tianhaowuhz, 2023-12-20:
It seems the problem is caused by some value in urdf is "e-40", which exceed the limit, after fix this, the model can be loaded.

tianhaowuhz, 2023-12-21:
Although I have been able to load the model, the retarget result from human video in example seems a little weird, for example, at the initial frame the finger of human hand is open, but the retargeted leap hand's finger is bent, is there any parameter I should try to adjust this?  
Here is my current yml config file of leap hand
`retargeting:
  type: vector
  urdf_path: leap_hand/leap_hand_right.urdf
  wrist_link_name: "base"

  # Target refers to the retargeting target, which is the robot hand
  target_joint_names: null
  target_origin_link_names: [ "base", "base", "base", "base" ]
  target_task_link_names: [ "thumb_fingertip", "fingertip", "fingertip_2", "fingertip_3" ]
  scaling_factor: 1.6

  # Source refers to the retargeting input, which usually corresponds to the human hand
  # The joint indices of human hand joint which corresponds to each link in the target_link_names
  target_link_human_indices: [ [ 0, 0, 0, 0 ], [ 4, 8, 12, 16 ] ]

  # A smaller alpha means stronger filtering, i.e. more smooth but also larger latency
  low_pass_alpha: 0.2
`

yzqin, 2023-12-21:
The `yml` configuration file seems to be in order; there don't appear to be any faults with it. The actual issue lies in the positioning of the `fingertip` link—it isn't accurately positioned at the fingertip of the leap hand as one might expect. The frame definition for the `fingertip` link is presented below:

[image]

yzqin, 2023-12-21:
The solution is to add a finger tip virtual link for leap hand so that the retargeting computation can be performed easily. 

Please check the recent commit 3aa3e1a3345ef8f6052792299538e7513f63911d. We provide a `yml` config file for the leap hand for both vector retargeting and dexpilot retargeting. You can use our provided model in this repo.

yzqin, 2023-12-21:
Here is the video generated based on [DexPilot config of leap hand](https://github.com/dexsuite/dex-retargeting/blob/main/dex_retargeting/configs/teleop/leap_hand_right_dexpilot.yml)

https://github.com/dexsuite/dex-retargeting/assets/24649446/bc26c53d-f2f4-424f-99a5-df8a54cd972b

tianhaowuhz, 2023-12-21:
It do looks better, thank you!

closed 2023-12-21 by tianhaowuhz