dexsuite/dex-retargeting #9: How to relax restriction on collision between links?

RyanJiao, 2024-01-01:
Hi, I was trying the leap hand urdf in the dex-urdf repository. And I manually add an fingertip link like this：
``` <link name="RF_touchtip" />
  <joint name="RF_fixtip" type="fixed">
      <parent link="fingertip_3" />
      <child link="RF_touchtip" />
      <origin rpy="0.0 -1.57079633 0.0" xyz="-0.01 -0.04 0.01 " />
  </joint>
```
And this is the yaml I used:
```
retargeting:
  type: vector
  urdf_path: yzqin_leap_hand/leap_hand_right.urdf
  wrist_link_name: "base"

  # Target refers to the retargeting target, which is the robot hand
  target_joint_names: null
  target_origin_link_names: [ "base", "base", "base", "base" ]
  target_task_link_names: [ "TH_touchtip", "FF_touchtip", "MF_touchtip", "RF_touchtip" ]
  scaling_factor: 1.4

  # Source refers to the retargeting input, which usually corresponds to the human hand
  # The joint indices of human hand joint which corresponds to each link in the target_link_names
  target_link_human_indices: [ [ 0, 0, 0, 0 ], [ 4, 8, 12, 16 ] ]

  # A smaller alpha means stronger filtering, i.e. more smooth but also larger latency
  low_pass_alpha: 0.2
```
The pose seems to be alright, but when I try to grasp something, the gap between fingertips looks too huge. I add a video to show the problem. And I' like to ask how to relax restriction on collision between links?
https://github.com/dexsuite/dex-retargeting/assets/56286149/7441fd18-96b5-4d9e-a037-2e25b9d4e0cc

RyanJiao, 2024-01-01:
[1-1-out.webm](https://github.com/dexsuite/dex-retargeting/assets/56286149/21ffdaf0-1056-4f32-949a-0b376c713c01)

RyanJiao, 2024-01-01:
[image]

yzqin, 2024-01-03:
Hi @RyanJiao 

The video you uploaded seems not functional. Could you re-upload a new one or in different format.

For grasping applications, `DexPilot` retargeting method is preferred over `vector` retargeting. Checkout the config [here](https://github.com/dexsuite/dex-retargeting/blob/main/dex_retargeting/configs/teleop/leap_hand_right_dexpilot.yml) for leap hand.

By the way, I was wondering that you we add the fingertip links by yourself. The leap hand model provided in this repo already has finger tip added, like in the config file above.

RyanJiao, 2024-01-03:
Hi Yuzhe,
Thank you for your reply, I tried the `DexPilot` and indeed the hand could actually grasp now.
Sorry for the silly questions.
And here is how I add the extra fingertip links, I wonder if this will work better?
[image]
[image]

yzqin, 2024-01-03:
Hi @RyanJiao 

No worries and I may need to write a summary about when to use different types of retargeting method. By the way, you modified URDF looks great! Thanks for letting me know!

RyanJiao, 2024-01-03:
Thank you again, I will close this issue now.

closed 2024-01-03 by RyanJiao