dexsuite/dex-retargeting #11: Plan to add dummy joints under Pin Model?

ariszxxu, 2024-04-20:
Hi Yuzhe @yzqin 

Thanks for your great work! I noticed that you tried to remove requirements on sapien in your last few commits and replaced them with pinocchio models. It's really thoughtful. 👍

However, the support of dummy joints may have also been removed, which may be necessary for position-based retargeting from my perspective. Do I understand this correctly? If so, will dummy joints be added under the current framework in the future?

Thanks again. 🙏

yzqin, 2024-05-17:
Hi @ariszxxu,

Thank you for your keen observation! It seems you've thoroughly examined the codebase to notice this update.

I have some **good news**: support for both dummy joints and position retargeting has been reinstated in the updated, sapien-free retargeting library. Note that while the main library no longer depends on sapien, version 3.0.0b0 is still necessary for visualizing examples.

You can now utilize the library for position retargeting. Additionally, the support for new dummy joints has been integrated at the XML level, allowing for dynamic generation of new URDF files with dummy joints during initialization. `pip install dex_retargeting==0.3.1` should get you here.

These updates are committed in c7692d0ec2ce8f36ee210c3bf581b04da8b89f2d and 59b044ea46ae361e76e72cd9f6e6c0d37dafec30

ariszxxu, 2024-05-19:
Thanks! @yzqin

closed 2024-05-19 by ariszxxu