dexsuite/dex-retargeting #13: question about mano conver

xbkaishui, 2024-05-27:
Hi 

Thanks for you greate work. I have some question about the vector retarget code. 
in  [single handle detector L 87](https://github.com/dexsuite/dex-retargeting/blob/main/example/vector_retargeting/single_hand_detector.py#L87),  the detect results are mapped to MANO convention,  what is the purpose of the convert ? can you give me more details ?

Thanks

yzqin, 2024-05-27:
Hi @xbkaishui 

Good question! 
Here's the rephrased GitHub issue reply in Markdown format:

## Wrist Frame Conventions

The code you're referring to maps the wrist frame convention used by MANO to the operator frame convention used throughout my repositories, including AnyTeleop and other teleoperation projects. This consistency across projects aims to simplify development and understanding.

**Operator Frame Convention:**

* **Right Hand:**
    * X-axis: Forward direction
    * Y-axis: Points from pinky to thumb
    * Z-axis: Points from wrist to fingertip

* **Left Hand:**
    * X-axis: Forward direction
    * Y-axis: Points from thumb to pinky (reversed compared to the right hand)
    * Z-axis: Points from wrist to fingertip

[image]

The reason for using the operator frame convention is to maintain consistency with the robot hand models in the [dex-urdf](https://github.com/dexsuite/dex-urdf) repository. 

Retargeting from human to robot hand poses is only possible if both use the same frame convention.  In other words, you can achieve accurate retargeting by ensuring the human and robot wrist conventions are aligned, even if they differ from the original MANO convention.

xbkaishui, 2024-05-28:
ok，got it Thanks

closed 2024-05-28 by xbkaishui