dexsuite/dex-retargeting #16: hand detection retarget not accurate

xbkaishui, 2024-05-28:
hello 

I have tested the hand detection retarget using allegro hand.  the test code is [ pr ](https://github.com/dexsuite/dex-retargeting/pull/15)

I found some position is not accurate, below are some test snapshots

 
[image]
[image]

### tested env:

- camera:  built-in camera on the computer is used.
- pc board name rog7
- system:  windows 11

yzqin, 2024-05-31:
Hi @xbkaishui 

This is a common issue of detection error. Luckily, the detection error can be mitigated by using a different retargeting algorithm.

I believe you are using the `RetargetingType.vector` as retargeting type, can you try with `RetargetingType.dexpilot`, I think it will solve the problem you mentioned above. 

By the way, thanks for the PR. I will review it and add it to the main repo with some small modifications.

yzqin, 2024-05-31:
Hi @xbkaishui 

I have merged your PR and updated the README for vector retargeting. I have also uploaded an example video captured on my own using the allegro hand with `dexpilot` retargeting config. Maybe you can take a look:

https://github.com/dexsuite/dex-retargeting/tree/main/example/vector_retargeting

xbkaishui, 2024-06-02:
> Hi @xbkaishui
>
> This is a common issue of detection error. Luckily, the detection error can be mitigated by using a different retargeting algorithm.
>
> I believe you are using the [...]

> Hi @xbkaishui
>
> This is a common issue of detection error. Luckily, the detection error can be mitigated by using a different retargeting algorithm.
>
> I believe you are using the [...]

hello， sorry to bother you again. i use dexpilot.  after some debug, I found that it is the mediapipe detection algo not so accurate. thanks for your response

xbkaishui, 2024-06-02:
> Hi @xbkaishui
>
> I have merged your PR and updated the README for vector retargeting. I have also uploaded an example video captured on my own using the allegro hand with `dexpilot` retargeting [...]

> Hi @xbkaishui
>
> I have merged your PR and updated the README for vector retargeting. I have also uploaded an example video captured on my own using the allegro hand with `dexpilot` retargeting [...]

thanks

closed 2024-06-02 by xbkaishui

BIT-wh, 2024-06-11:
> hello
>
> I have tested the hand detection retarget using allegro hand. the test code is [ pr ](https://github.com/dexsuite/dex-retargeting/pull/15)
>
> I found some position is not accurate, below are [...]

Hi, sorry to bother you. I have a problem in install dex-retargeting in Windows11. When I install it, it makes a mistake in install pin. Then I use `conda install pinocchio -c conda-forge` to install. It works, but when I use `pip install dex_retargeting`, It still downloads` /pin-2.7.0.tar.gz` and then reports an error, as before. Bro, how did you download it in win11?  Looking forward to your reply, thank you !

xbkaishui, 2024-06-18:
> Hi, sorry to bother you. I have a problem in install dex-retargeting in Windows11. When I install it, it makes a mistake in install pin. Then I use `conda install pinocchio -c conda-forge` to [...]

use conda install 

conda install pinocchio -c conda-forge