dexsuite/dex-retargeting #19: Import the new manipulator's URDF

Jack-XingLi, 2024-06-07:
Hi @yzqin ，
Your work has left a deep impression on me, but I am currently encountering issues while using the code. I am attempting to import a humanoid three-finger dexterous hand, consisting of a thumb, index finger, and middle finger. I have imported the URDF, but it seems that the redirection process has encountered some problems. I would like to inquire if there are any specific considerations to keep in mind when importing URDF files. I look forward to your response. Thank you!
[image]
When my humanoid hand is fully extended, the robotic hand does not synchronize with mine; instead, it bends to 90 degrees.

yzqin, 2024-06-07:
hi @Jack-XingLi 

Thanks for the kind words. Yes there are some specific considerations for the hand definition. To use dex-retargeting, we assume that the root of your hand should be consistent with some conventions.

Right hand: x->forward, z->from wrist to tip, y->pinky to thumb
Left hand: x->forward, z->from wrist to tip, y-> thumb to pinky

An visualization can be found below. 

[image]

I have made some effort to ensure that all models we used in our repo, such as [dex-urdf](https://github.com/dexsuite/dex-urdf), [bunny_vision_pro](https://github.com/Dingry/BunnyVisionPro), and the customized robot hand design in [from-one-hand-to-multiple-hand](https://github.com/yzqin/dex-hand-teleop), will follow this convention. So that if you use any hands from dex-urdf, it will work out of the box. For your new hand, you may also need to rotate the urdf so that it follows the same convention.

By the way, may I ask what is the name of your humanoid robot hand?

Jack-XingLi, 2024-06-07:
hi, @yzqin :
Thank you for your prompt reply. This hand is a prototype made in our lab, very similar to the Leaphand, except it lacks a ring finger, and the thumb has only 3 degrees of freedom. I've adjusted the base coordinate system to meet specific requirements, but it seems I haven't successfully resolved my issue. Below is how my hand looks when extended, but after redirection, the robotic hand tends to bend to 90 degrees, while my hand is extended. I suspect it's an issue with the redirection coefficient, but despite many adjustments, I haven't been able to solve the problem effectively. Do you have any suggestions? Thank you!
[image]
[image]
[image]
[image]

Jack-XingLi, 2024-06-09:
I have solved the problem, the main reason was the URDF itself, thank you for your work!

yzqin, 2024-06-10:
Glad to hear that you were able to resolve the problem! I apologize for not being able to provide more assistance in this case. Without a complete URDF file, it was a bit challenging for me to fully grasp the context of the issue you were experiencing.

Since the problem has been solved, I will go ahead and close this issue. If you encounter any further difficulties or have additional questions, please don't hesitate to open a new issue with more detailed information, including the relevant URDF file if possible.

closed 2024-06-10 by yzqin

boreep, 2025-02-20:
> hi, [@yzqin](https://github.com/yzqin) : Thank you for your prompt reply. This hand is a prototype made in our lab, very similar to the Leaphand, except it lacks a ring finger, and the thumb has [...]

Hello, I would like to know how you got the eligible URDF file