dexsuite/dex-retargeting #33: Basic Mathmatics about calculating the joint_pos variable in SingleHandDetector.py

CodingCatMountain, 2024-08-10:
Hi, @yzqin. Thanks for sharing a such good job! I have noticed that dex-retargeting is such useful, and a few tele-operation jobs, (e.g., bunny-visionpro, avp-teleoperate) all choose to build upon it. And I hope you can help me understand the basic mathmatics about calculating the joint_pos variable in SingleHandDetector.py. In the line 87 of this script, I can't figure out why the joint_pos is calculated like this:

```
keypoint_3d_array = self.parse_keypoint_3d(keypoint_3d)
keypoint_3d_array = keypoint_3d_array - keypoint_3d_array[0:1, :] # calculate the relative positions of each finger
mediapipe_wrist_rot = self.estimate_frame_from_hand_points(keypoint_3d_array) # calculate the orientation of wrist
joint_pos = keypoint_3d_array @ mediapipe_wrist_rot @ self.operator2mano # Why????
```
And why the self.operator2mano is :

```
OPERATOR2MANO_RIGHT = np.array(
    [
        [0, 0, -1],
        [-1, 0, 0],
        [0, 1, 0],
    ]
) # Why??
```
Looking forward to your reply!

yzqin, 2024-08-13:
Hi @CodingCatMountain 

To use retarget human data to robot data, we need to define a canonical frame where the robot orientation matches that of humans. You can check [this issue](https://github.com/dexsuite/dex-retargeting/issues/13#issuecomment-2133886267) for more information.

Basically, these two lines of code convert the detected results from mediapipe (which uses MANO convention) to our convention mentioned above.

CodingCatMountain, 2024-08-14:
@yzqin Cool! Now, I think i can understand these two lines. I will close this issue.

closed 2024-08-14 by CodingCatMountain