dexsuite/dex-retargeting #42: Retargeting is different in real life and simulation.

kaijieshi7, 2024-10-05:
Hi, thanks for your excellent work.

Currently, I'm trying to retarget my hand to ability hand in real life. However, I find that even though the retargeting in simulation is awesome, the retargeting in real life is different from simulation. As we can see in the attached video, when I close the thumb and index finger, they are close in simulation but still a distance in real life.
Could you give me suggestions about which part I should change to make a real-life retargeting successful?

https://github.com/user-attachments/assets/955e35db-a281-47cd-b310-4c6e935b088a

yzqin, 2024-11-13:
hi @kaijieshi7 

Maybe you can tune the PID parameters of your ability hand to make it stiffer. The control parameters seem not so strong. The issue is that the ability hand can not reach the position command you provided.

kaijieshi7, 2024-11-13:
Ok. Thank you very much.

closed 2024-11-13 by kaijieshi7

a-sword-no-afraid, 2025-01-17:
Can you tell me how to replace it with my own hand? I modified my urdf and wrote a yaml, but the effect is poor. thanks

Naveench7, 2025-07-09:
@kaijieshi7 i am working on the same problem with same ability hand, could you help me guide, what necessary changes needed to be made so that i can retarget to the real hand