dexsuite/dex-retargeting #43: Modifying Dataset (i.e. GRAB)

bb0928, 2024-11-10:
Hi, thanks for your project, excellent work. However, when I attempted to modify dataset (from Dex-YCB -> GRAB), I found that the rendered result of MANO hand in SAPIEN does not match the ground truth (the hand pose of GRAB should be).
When I was debugging, I've noticed that in "./example/position_retargeting/mano_layer.py", whose forward function calculates vertices and joints, the note shows that it is from "dexycb-toolkit". Since GRAB have their own function of doing this, I am now wondering if my issue is caused by incompatible function.
Many thanks.

yzqin, 2024-11-13:
Hi @bb0928,

I suggest creating a `grab_dataset.py` script to convert MANO format data into vertices and joints that SAPIEN can render. 

SAPIEN cannot interpret MANO parameters directly - it relies on `mano_layer.py` (as seen in the dexycb example) to generate meshes it can use.

For your issue, I recommend investigating two potential problems:

1. MANO Pose Parameter Normalization
   - Check if GRAB's MANO poses are normalized by mean pose
   - Some implementations use absolute axis angles
   - Others use relative angles (difference from mean male/female pose)
   - This distinction can cause errors

2. MANO vs SMPL-X Format
   - Verify whether GRAB uses original MANO or newer SMPL-X format
   - While SMPL-X uses the same number of parameters as MANO, it uses different linear bases
   - Using MANO functions to parse SMPL-X parameters won't raise errors but may produce incorrect results since only the shape dimensions match

bb0928, 2024-11-15:
thanks a lot, i've done so, and it really works. However, there is another question about sapien renderer. 
in your script "hand_viewer.py", there is a function
    def _update_hand(self, vertex):
        self.clear_node()
        normal = compute_smooth_shading_normal_np(vertex, self.mano_face)

        mesh = self.context.create_mesh_from_array(vertex, self.mano_face, normal)
        model = self.context.create_model([mesh], [self.mat_hand])
        node = self.internal_scene.add_node()
        node.set_position(np.array([0, 0, 0]))
        obj = self.internal_scene.add_object(model, node)
        obj.shading_mode = 0
        obj.cast_shadow = True
        obj.transparency = 0
        self.nodes.append(node)

I didn't quite understand what does "node" means, wondering if it is the global position or not?
I have this question because when dealing with the motion of GRAB object, i also update its vertices by step, rather than using Pose function defined in SAPIEN (i.e. quat+trans), by creating another __update_object method. However, the render result shows that the object didn't move at all.

I exported the HOI meshes by frame, visualized them in meshlab, and found that the vertices position of hand and object are correct. so i think this issue probably stems from the rendering process.

Could you please offer any help about this? thanks!

yzqin, 2024-11-15:
Node is a unit in a scene hierarchy. It is the same as "node" in three.js to define a tree structure. It is not a pose. 

I use this function to render mano hand because the mesh of MANO changes each step. I think you do not need to use this function to handle GRAB object since the object mesh is fixed. I recommend you use the standard Pose function defined in SAPIEN (i.e. quat+trans) to update object.

bb0928, 2024-11-16:
Thanks for your response.
I have used standard pose parameters in sapien at first, but found that there are slight incorrespondence between hand and object. Like the figure below:[image]

I guess it's because the 'center' of object is wrong so the actual trans and rot are not what i expected. So i wrote an object layer instead, and output the meshes by frame.
Btw, the official render code of GRAB is also update by vertices.

bb0928, 2024-11-16:
Prior to visualize the sequence in sapien, i exported the meshes and visualized them in meshlab to prove that the relative pose of hand and object is correct.
[image]
Then, i wrote a function _update_object  to imitate your function. But it seems that the object doesn't move at all during render process (i.e. remains at origin)
I'm wondering what can cause this issue.

bb0928, 2024-11-16:
I think i have understood the function and solve this issue. Thanks for your help.

closed 2024-11-16 by bb0928