dexsuite/dex-retargeting #47: steps to add a custom hand

zhongxiangzju, 2024-12-03:
Nice work! Would to please share the steps about how to add a custom hand and perform retargeting? Thanks!

yzqin, 2024-12-03:
@zhongxiangzju 

Creating a config file for a new robot hand is not so hard, but the tricky thing is that you need a hand and with good quality. Which hand do you want to add? Do you have a link to the manufacture provided urdf?

zhongxiangzju, 2024-12-04:
Thanks for quick reply! I have added urdf of the custom dexterous gripper and run position retargeting. The results seem reasonable but with serious piercing problems. 
[image]

Any suggestions to improve the results? Thanks.

zhongxiangzju, 2024-12-04:
And could you please introduce the difference between position retargeting and vector retargeting?

yzqin, 2024-12-05:
To put it simply, vector/DexPilot retargeting is mainly for teleoperation while position retargeting is for retargeting from video

yzqin, 2024-12-05:
To alleviate the penetration, you can try to modify the retargeting config and add more position on the robot hand

1024AILab, 2024-12-07:
@zhongxiangzju Hi bro, how to create a config file on custom hand ? Thank you~