dexsuite/dex-retargeting #51: Use of sapien on a Nvidia board

cixliv, 2025-02-05:
Hey there, I have been trying to get the https://github.com/unitreerobotics/avp_teleoperate with dex-retargeting to work on an Nvidia NX board.

The dex-retargeting version it is using 0.1.1 requires sapien to run, sapien does not work on an ARM based board.

Do you recommend updating to a newer version of dex retargeting? or do you recommend compiling SAPIEN in a way that runs on the Nvidia board?

yzqin, 2025-02-05:
The latest dex-retargeting does not relies on sapien, check the [setup.py](https://github.com/dexsuite/dex-retargeting/blob/main/setup.py). In the current version, I only use the SAPIEN in the example for rendering and visualization in simulator. But it is not necessary if you want to teleop the real robot. 

I do not know why unitree is using an old version of dex-retargeting but for their application, the new dex-retargeting should also work.

cixliv, 2025-02-05:
This is the file that is being used https://github.com/unitreerobotics/avp_teleoperate/blob/h1_2/teleop/inspire_hand.yml

That likely needs to changed to work with 0.2 and above. Potentially notable that I’m using an Nvidia NX board.

Thanks!

a-sword-no-afraid, 2025-02-06:
> The latest dex-retargeting does not relies on sapien, check the [setup.py](https://github.com/dexsuite/dex-retargeting/blob/main/setup.py). In the current version, I only use the SAPIEN in the [...]

Hello, if I use my own URDF for retargeting, do I need to keep the coordinate system consistent with the URDF you provided? Because when I replaced it with my own URDF, it seems that the mapping went wrong.

cixliv, 2025-02-06:
Here is more info from unitree on why they are using v0.1.1.1 

After updated it will now mean the the joints need to be updated 

https://github.com/unitreerobotics/avp_teleoperate/issues/30

yzqin, 2025-02-07:
@a-sword-no-afraid 

Yes, you need to keep the coordinate system consistent with the URDF provided. The hand xyz frame should follow this convention: x-axis face forward, z axis from wrist to finger tip. For right hand, y-axis is from wrist to thumb, for left hand, y-axis is from thumb to wrist. 

[image]

samcts-auki, 2025-02-24:
did you eventually get it to work? I am also doing the same with G1_29 with inspire hands. The only error left is `Mimic joint should not be one of the target joints.`