dexsuite/dex-retargeting #53: how to grasp in spaien simulator?

longshot-pting, 2025-03-13:
Hi, I want to grasp a cube in simulator, but I found the root can not move when my hand root was moving. Can you give me some suggestions about how can I achieve this goal? I will be appreciate if you help!

yzqin, 2025-03-13:
For the wrist motion and how to freely move a hand, you can check our previous effort here: https://github.com/yzqin/dex-hand-teleop/tree/main

It is also based on the same simulator and follow the same convention for the hand retargeting.

longshot-pting, 2025-03-14:
> For the wrist motion and how to freely move a hand, you can check our previous effort here: https://github.com/yzqin/dex-hand-teleop/tree/main
> 
> It is also based on the same simulator and follow the same convention for the hand retargeting.

Thank you very much, I will try it as soon so possible

closed 2025-03-25 by yzqin