dexsuite/dex-retargeting #61: Problems with the movement of the robotic arm

anonymouseea, 2025-04-14:
[image]
I have realized, the redirection of the manipulator, such as and the realization of the compound movement of adding the manipulator to the end of the manipulator, I noticed that the assembly in assets has a robot model that has been assembled, how to use it, can you give me the approximate steps.Looking forward to hearing from you!

yzqin, 2025-04-14:
Sorry I do not quiet get you. What do you mean how the robot model has been assembled? Are you looking for URDF combining hand and arms? If it is the case, you can try: https://github.com/dexsuite/dex-urdf

anonymouseea, 2025-04-15:
> Sorry I do not quiet get you. What do you mean how the robot model has been assembled? Are you looking for URDF combining hand and arms? If it is the case, you can try: https://github.com/dexsuite/dex-urdf

What I mean is that I currently have dex-urdf, but I'm not sure how to use it. When I run show-realtime-retargeting.py, there is only one hand involved, without a robotic arm. If I want to incorporate the movement of a robotic arm, roughly what steps should I take? As you can see from the example picture, it is with a robotic arm 😭

anonymouseea, 2025-04-16:
是否有现成的带有机械臂和手部的代码，我自己尝试将ur5e_shadow_hand.urdf加入到模拟环境中，但机械臂并不会运动，求助！

closed 2025-07-10 by yzqin