dexsuite/dex-retargeting #62: arm control

anonymouseea, 2025-04-22:
Are there any ready-made codes with robotic arms and hands? I tried to integrate the ur5e_shadow_hand.urdf into the simulation environment and wrote the corresponding code myself, but as you can see, it doesn't perform very well. I noticed that you used CUROBO in your paper. Are there any related implementation programs? If not, could you provide a general idea? Help!

https://github.com/user-attachments/assets/0546a413-4d5b-4c05-afbd-db730c036b40

yzqin, 2025-04-22:
You can take a look of the https://github.com/Dingry/BunnyVisionPro. It has arm control examples.

anonymouseea, 2025-04-23:
> You can take a look of the https://github.com/Dingry/BunnyVisionPro. It has arm control examples.
Thank you very much！！！

yzqin, 2025-07-05:
@Jason3900 Yes, this will definitely be an issue. If pred_joints_smpl consistently shows wrist poses of [0,0,0], there are two likely causes:

1. You're using a local hand pose estimator (like MediaPipe) that doesn't track wrist orientation relative to the global coordinate system
2. You're using a global hand pose estimator but inadvertently converting the output to local coordinates through your transformation pipeline

Since the FrankMocap detector used in dex-hand-teleop is designed as a global pose estimator, it should be capable of detecting the actual global wrist pose. I'd recommend checking your coordinate transformation steps to ensure you're preserving the global wrist orientation data.

Jason3900, 2025-07-07:
@yzqin Thanks, I find a global hand pose estimator and make it work on wrist movement in Isaac Gym. But I'm struggling with the finger movement. I tried to scale up/down values of the finger position (25x3). Can you give me some advice? Specifically, I replace the Television object to take a GRB camera/video as input, estimate the hand pose by a global MANO-style hand estimator, return world-coordinates of the hand and wrist position in 21x3 with a rotation matrix 3x3. I interpolate the metacarpal 4 points manually, create the hands var with 4x4 matrix (3x3 rotation matrix with hand keypoints[0] which is (3,)) then convert all the basis to meet Open-XR's convention. I can confirm that the wrist movement is just as I expected in the Isaac Gym with inspire hand. But the finger movement is too small to visualize, even though the numbers are changing. 
Looking forward to your reply.

yzqin, 2025-07-10:
Different hand tracking estimators use different conventions for representing hand poses. The current examples in dex-retargeting follow the MediaPipe convention. To adapt OpenXR data to work with our system, you'll need to convert from the OpenXR 25-joint format to MediaPipe's 21-joint format by excluding the metacarpal joints, as detailed in the [Godot OpenXR hand tracking documentation](https://docs.godot.community/tutorials/xr/openxr_hand_tracking.html).

Additionally, you may need to adjust the wrist orientation definition to align with the coordinate system used in this repository. This is the **most tricky** part since dex-retargeting use the local (with respect to wrist) position of each finger position for retargeting, so that you need to make sure that the wrist orientation of your detector aligns with the robot.

Simease, 2025-08-20:
> 谢谢，我找到了一个全局手部姿势估计器，并让它在 Isaac Gym 中用于手腕运动。但我在手指运动方面遇到了困难。我尝试放大/缩小手指位置的值 （25x3）。你能给我一些建议吗？具体来说，我将 Television 对象替换为 GRB 摄像机/视频作为输入，通过全局 MANO 风格的手部估计器估计手部姿势，使用旋转矩阵 3x3 返回 21x3 的手和手腕位置的世界坐标。我手动插值掌骨 4 点，使用 4x4 矩阵创建手 var（带有手关键点[0] 的 3x3 旋转矩阵，即 （3，）），然后转换所有基以满足 Open-XR 的约定。我可以确认手腕运动正如我在 Isaac Gym 中用 Inspire 手所期望的那样。但是手指的移动太小而无法想象，即使数字在变化。期待您的回复。

Could you please share the global hand pose estimator you found? It would be really helpful for my work as well. Thanks a lot!